nasa/gunns #100: Address Sanitizer Error

genti337, 2024-01-05:
Found an error in GUNNS when running the address sanitizer one of our sims that uses GUNNS.  The issue can be duplicated by adding the following flags to the `S_overrides.mk`.

- TRICK_CFLAGS   += -g  -fstack-protector-all -fsanitize=address
- TRICK_CXXFLAGS += -g  fstack-protector-all -fsanitize=address

This results in the follow error.

```
|L   3|2024/01/05,07:18:40|JSLAL0522090204| |T 0|0.000000| H&S: unable to open '../../../logs/TS_Health_and_Status.out' for appending=================================================================
==41364==ERROR: AddressSanitizer: stack-use-after-scope on address 0x00016f067b60 at pc 0x000104ec5d0c bp 0x00016f067950 sp 0x00016f067110
READ of size 6 at 0x00016f067b60 thread T0
    #0 0x104ec5d08 in wrap_strlen+0x264 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x15d08)
    #1 0x100db1f60 in std::__1::__constexpr_strlen[abi:v160006](char const*) cstring:114
    #2 0x100db01cc in std::__1::char_traits<char>::length(char const*) char_traits.h:220
    #3 0x100df5d54 in std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string[abi:v160006]<std::nullptr_t>(char const*) string:882
    #4 0x100d9e874 in std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string[abi:v160006]<std::nullptr_t>(char const*) string:880
    #5 0x100e9d4d0 in tsHsSendMsgFileLine(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, unsigned int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, TsHsMsg const&) TsHsMsg.cpp:370
    #6 0x100f4c1a4 in Gunns::initialize(GunnsConfigData const&, std::__1::vector<GunnsBasicLink*, std::__1::allocator<GunnsBasicLink*>>&) Gunns.cpp:481
    #7 0x100e8dddc in TestFluidNetwork::initNetwork() TestFluidNetwork.cpp:159
    #8 0x100f0ac58 in GunnsNetworkBase::initialize(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) GunnsNetworkBase.cpp:116
    #9 0x100d9de14 in TestSimObject::call_function(Trick::JobData*) S_source.cpp:1090
    #10 0x101eb6830 in Trick::JobData::call() JobData.cpp:182
    #11 0x101dea354 in Trick::Executive::call_initialization() Executive_call_initialization.cpp:26
    #12 0x101df45d4 in Trick::Executive::init() Executive_init.cpp:45
    #13 0x101f967dc in master(int, char**) master.cpp:42
    #14 0x100da6c04 in main S_source.cpp:1502
    #15 0x1a288ff24  (<unknown module>)

Address 0x00016f067b60 is located in stack of thread T0 at offset 96 in frame
    #0 0x100e9d1e8 in tsHsSendMsgFileLine(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, unsigned int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, TsHsMsg const&) TsHsMsg.cpp:360

  This frame has 4 object(s):
    [32, 56) 'mtext' (line 364)
    [96, 120) 'ref.tmp' (line 368) <== Memory access at offset 96 is inside this variable
    [160, 184) 'ref.tmp7' (line 370)
    [224, 248) 'ref.tmp11' (line 370)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-use-after-scope (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x15d08) in wrap_strlen+0x264
Shadow bytes around the buggy address:
  0x00016f067880: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067900: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067980: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 f8 f2 f8 f3
  0x00016f067a00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067a80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x00016f067b00: f1 f1 f1 f1 00 00 00 f2 f2 f2 f2 f2[f8]f8 f8 f2
  0x00016f067b80: f2 f2 f2 f2 00 00 00 f2 f2 f2 f2 f2 f8 f8 f8 f3
  0x00016f067c00: f3 f3 f3 f3 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067c80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067d00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x00016f067d80: f1 f1 f1 f1 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==41364==ABORTING
Process terminated by signal SIGABRT
```

Tracing this back, found that changing the variable `subsys` on line 368 of `TsHsMsg.cpp` from a `const char*` to `std::string` fixes the error.

excaliburtb, 2024-01-05:
not to pile on but here's a suggested change via git diff:

```
--- a/ms-utils/software/SimCompatibility/TsSimCompatibility.hh
+++ b/ms-utils/software/SimCompatibility/TsSimCompatibility.hh
@@ -527,7 +527,10 @@ class TestHookSimObj;
 ////////////////////////////////////////////////////////////////////////////////////////////////////
 #ifndef no_TRICK_ENV
 #define TS_DELETE_ARRAY(ptr) \
-    TsMemoryManager::tsDelete(reinterpret_cast<void*>(ptr), typeid(ptr).name())
+    { \
+       TsMemoryManager::tsDelete(reinterpret_cast<void*>(ptr), typeid(ptr).name()); \
+       ptr = 0; \
+    }
 #else  // non-Trick
 #define TS_DELETE_ARRAY(ptr) \
     { \

```

jasonlharvey, 2024-01-12:
To build with asan, need to also add **-lasan** to _TRICK_USER_LINK_LIBS_, and install **libasan** on rh8.

closed 2024-01-19 by jasonlharvey