nasa/gunns #137: Update ROSES to work with latest CUDA

jasonlharvey, 2024-11-07:
A user reports that the ROSES stuff doesn't compile with CUDA 12.2, in RHEL8, RTX4090 card.  Something about the sparse/dense matrix conversion function calls no longer match the CUDA API, possibly the cusparseDdense2csr call in CudaSparseSolve::Decompose.  It's been a while since we implemented ROSES, with CUDA 8.0, and we should step up to the latest now.

jasonlharvey, 2025-01-24:
cuSPARSE v8 documentation for cusparseDdense2csr: https://docs.nvidia.com/cuda/archive/8.0/cusparse/index.html#cusparse-lt-t-gt-dense2csr

cuSPARSE latest (v12) docs for cusparseDpruneDense2cdr, which appears to be the closest to the legacy cusparseDdense2csr, however note that they say this is deprecated: https://docs.nvidia.com/cuda/cusparse/#cusparsexprunedense2csr-deprecated

Looks like somewhere between 8 and 12 they changed all the dense to sparse format conversions to add the prune feature, which omits values in the dense matrix that are smaller than a given threshold from the sparse matrix result.  This would have been a useful feature, but all the dense to sparse matrix conversions appear to be deprecated anyway.  To avoid another future break due to this deprecation, we might ought to do our own conversion to the sparse format, and not use CUDA for the conversion, or find out if there's a proper way to do it with current & future CUDA versions.

jasonlharvey, 2025-01-24:
More deprecated functions we are currently using that need to be future-proofed:

cusolverSpDcsrlsvchol, which we use for our sparse matrix Cholesky: https://docs.nvidia.com/cuda/cusolver/index.html#cusolversp-t-csrlsvchol-deprecated

jasonlharvey, 2025-01-24:
More changes & incompatibilities:

* The type _cusparseStatus_t_ that we use in _CudaIfUtils::getCudaReturnString_ and elsewhere, has its set of values changed.  Values CUSPARSE_STATUS_MAPPING_ERROR, CUSPARSE_STATUS_ZERO_PIVOT no longer exist, and instead there are new values CUSPARSE_STATUS_NOT_SUPPORTED, CUSPARSE_STATUS_INSUFFICIENT_RESOURCES.
* The type _cusolverStatus_t_ that we use in _CudaIfUtils::getCudaReturnString_ and elsewhere, has its set of values changed.  Values CUSOLVER_STATUS_MAPPING_ERROR, CUSOLVER_STATUS_ZERO_PIVOT, CUSOLVER_STATUS_INVALID_LICENSE no longer exist.
* Function _cudaGetErrorString_ that we use in _CudaIfUtils::checkReturn_, might no longer exist.  We probably need a new function, similar to our _CudaIfUtils::getCudaReturnString_ to convert the _cudaError_t_ values: https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__TYPES.html#group__CUDART__TYPES_1g3f51e3575c2178246db0a94a430e0038

jasonlharvey, 2025-01-28:
Notes on updates in commit 1832d47:

* Cleaned up the makefiles to remove specific CUDA version, i.e. -L/usr/local/cuda/lib64 instead of -L/usr/local/cuda-8.5/...
* Simplified the functions to report _cudaSolver_t_ and _cudaSparse_t_ status to H&S to just put the enum code as integer instead of trying to convert to string - the enum definitions in CUDA are changing too much in CUDA to keep up.  Users can refer to the CUDA API documentation to decipher the values.
* I went with the _cusparseDpruneDense2csr_ functions instead of _cusparseDenseToSparse_, even though the former are deprecated and the latter are not, because the latter use the wrong matrix descriptor _cusparseSpMatDescr_t_ instead of the _cusparseMatDescr_t_ that _cusolverSpDcsrlsvchol_ needs, and the CUDA documentation doesn't clearly describe the difference between those two matrix descriptor types and if they're interchangeable.  I expect these deprecated 'prune' functions that we're using now will go away in CUDA 13.  CUDA documentation says that the upcoming cuDSS library should be used for sparse matrix solving, so we'll have to rework this code to use it.  Current fix is a band-aid for 12.X, and I'll have to revisit in 13 or when i can get cuDSS to work with.
* ROSES unit tests are passing in manual runs with the ROSES option enabled, so I think the new code is working.  Note our GitHub CI doesn't run the ROSES tests.
* I don't know how these new 'prune' functions compare to the old functions for speed.  I haven't run the timing metrics sim yet and don't have a system on which i can A/B compare the two versions.

KatuunuXVI, 2025-02-04:
The file `GunnsElectConverterOutput.cpp` is missing the definition for the `getConverterEfficiency`. The repo currently returns a linking error. 
```
double GunnsElectConverterOutput::getConverterEfficiency() const {
    return mConverterEfficiency;
}
```

closed 2025-02-25 by jasonlharvey