nasa/gunns #146: Modify Dockerfiles to work around CentOS EOL changes

elizabeth-lynch, 2025-03-11:
Since CentOS EOL, mirrorlist and mirror.centos.org are no longer available.  Need to modify the Dockerfiles to point to vault.centos.org, instead.

jasonlharvey, 2025-03-14:
I don't see where we're using mirror sites in our Dockerfiles.  Is this FROM statement using one by default?

https://github.com/nasa/gunns/blob/05c743978f3928c0779c0e354919ae8680011a08/docker/unloaded/Dockerfile#L5

I don't run CI on the Centos Dockerfiles anymore.  At this point I'm wondering if we should just delete them.

Can you be more specific, or submit a pull request, with the changes you'd like?  Thanks!

elizabeth-lynch, 2025-03-14:
Yes, the FROM centos:7 uses it by default.  I have no objection to deleting the CentOS Dockerfiles if they are no longer being used.  I tried them first because they seemed to be the default.  If there is a desire to keep using them, I can submit a pull request with the fix.

jasonlharvey, 2025-03-21:
Ok after asking around, I'm going to delete the CentOS files.  Thanks for bringing this to my attention!

closed 2025-03-21 by jasonlharvey