nasa/gunns #149: Question on GunnsFluidSensor object GUNNS update v19.5

mc1998, 2025-04-28:
Hello all, may I request a bit more information on the update regarding the fluid sensor object. Gunns is automatically updating this component in my schematic, specifically the Ports section by adding an 'amp'. 

Since I am only using node 0, is this an update for drawIO/Gunns, revealing that node 1 is connected to the ground node (empty)? What does apm stand for?

Thanks!

jasonlharvey, 2025-04-29:
I was able to recreate this - running the netexport.py script causes the ```&``` symbol in the Ports field to be changed to ```&amp;``` which is the html code for ```&```.  I don't know whether this is netexport.py doing it or some internal draw.io feature yet.  Once you export the drawing and sync draw.io back to the modified drawing file, that ```&amp;``` will show up in the Ports field of the sensor's shape data.

That Ports field is basically just a comment and has no effect on the operation of the model.  The Ports field in the shape data is meant to be a quick reference to the user for what all the ports mean.  So in this case we can live with this problem for now since it shouldn't affect the model operation (let me know if you find otherwise).

I'll probably use this issue to update the default shape to change that problematic ```&``` to just say '0=node and +P' instead.

In the case of GunnsFluidValve, port 0 is the node you are sensing, and port 1 is the reference node.  For an absolute pressure sensor, the reference pressure is zero, so we connect port 1 to the Ground node (which has zero pressure).  But since that is a common use case and having all those ground nodes everywhere would clutter the drawing, we designed some links, like GunnsFluidSensor, to be able to default some of their ports to Ground if they're not drawn.  In this case, for your absolute pressure sensors you can leave the port 1 connection off the drawing and this will de-clutter the drawing.  The exported network will internally connect port 1 to Ground, making the sensor sense absolute pressure of the node at port 0.  If you have a delta-pressure sensor, then in your drawing you'd connect port 1 to a regular non-Ground node that is the reference pressure.  This is all illustrated in the link help page for the sensor in the wiki: https://github.com/nasa/gunns/wiki/GunnsFluidSensor#how-to-use-in-gunnsdraw

Thanks for bringing this issue to my attention!  I will work it as a valid bug.  In the meantime, you should be able to use your network even with that ```&amp;``` text in there.

mc1998, 2025-04-29:
Thank you for the quick response @jasonlharvey ! I'm glad I could make a difference :). I've already verified that this doesn't impact our sim runs. So there is no immediate action needed.