nasa/gunns #150: Question on Gunns Trickification

mc1998, 2025-04-30:
Hello, I'm currently exploring TRICK's trickify functionality for both JEOD & GUNNS and am running into a few build errors for Gunns. Below is my trail.

I see in `gunns/lib/trick_if/Makefile` that the trickify.mk is called. However, I cannot seem to find the `S_source.hh` file that is needed for trickification. Leading me to the `Makefile` found in `gunns/lib/trick` that will generate this file.

This makefile states that running `make`, "builds the libgunns.a library". Which is exactly what I want, as trickification requires the *.a and *.o libraries. However running this file results in the following error: 
```
~/directory/to/gunns/aspects/dynamics/GunnsDynSingleRigidEom.cpp:183:10: fatal error: sim_services/Integrator/include/integrator_c_intf.h: No such file or directory
 #include "sim_services/Integrator/include/integrator_c_intf.h"
```
This header file does exist and is found in Trick (19.7.3) - `~/trick/include/trick/compat/sim_services/Integrator/include/integrator_c_intf.h`. I have tried commenting it out and it results in more errors. I also tried running the SIM_class_test_compile to see if anything else broke, and I got this error.

[image]

Some background that may be beneficial: My main focus is to trickify jeod, which I have successfully achieved. However, this results in any non-jeod model to rebuild with trick-CP, regardless of present changes. So my current thinking is that my trickification of jeod is inadvertently causing Gunns trickification process to occur as well, as in my output it is the first to recompile, thus leading to this post. My current thinking is if gunns is failing to trickify (as seen above) then it is reverting to the regular gunns recompile everytime. This seems like a stretch as I would expect it to just crash throughout the build process. Any info relating to the correct procedure to trickify gunns will be beneficial! 

Thank you in advance.

mc1998, 2025-05-01:
After closer inspection I realize that my sim rebuilding after trickifying is not a result of Gunns. We can ignore the top half of my question above. I would still like to know how gunns implements trickification and if there is a resource somewhere for it.

jasonlharvey, 2025-05-02:
As you found, the _gunns/lib/trick_if_ depends first on building _gunns/lib/trick_, which generates the .o, .a, and S_sources.hh needed by _trick_if_.

This Trick wiki article is a reference on how to Trickify: https://nasa.github.io/trick/documentation/building_a_simulation/Trickified-Project-Libraries
It explains why we have the separate _trick_ lib as a precursor to the _trick_if_ lib and how they fit together.

Typically, trickified libs are optional in our internal NASA projects -- we'll set up our Trick sim makefile (S_overrides.mk, etc.) to look for a given lib, and if not found, fall back to Trick compiling that code in the sim build, as needed by make dependencies.  This sounds like what might be hapenning in your sim, since you said _"My current thinking is if gunns is failing to trickify (as seen above) then it is reverting to the regular gunns recompile everytime."_  That is a deliberate choice in the makefile setup, to fall back on compiling the code if the lib isn't found.  We make it optional in our projects for flexibility - it allows the same project to support using the trickfied libs for speed, and to not use them if we're working on a lot of changes to the code used in those libs instead.  Another benefit of not using the trickfied lib is that it can lead to a smaller executable size, since you're only including the code actually used in the sim, and not including the whole library.  If you'd rather abort the sim when the trickified lib couldn't be found, then adjust the Trick sim makefile to do that accordingly.  Another option might be to continue to build the sim, but output a glaring warning that the desired trickfied lib couldn't be found and needs investigation.

Did this answer your question?  Let me know if you have more questions or want to discuss more...thanks!

mc1998, 2025-05-02:
Thanks for the response, my problem is a bit different but your clarification does clear up a few things. Aside from gunns failing to build the libgunns.a library, my real issue is similar to what you mentioned.

The problem is that `make` is trying to link my gunns models to the trickified Jeod library (.o & .a). These links shouldnt exist/occur and thus `make` decides to recompile them every time. Where as before, gunns would only recompile if a gunns component changed or one of its dependencies.

Ill explore the trick github repo to see if there are any relevant posts. It may due to how I'm setting up my S_overrides.mk that is causing the issue.

`Prerequisite '/dir/to/jeod/trickified/trickified_jeod_Linux_7.5_x86_64.o' of target 'build/dir/to/gunns/aspects/electrical/Batt/GunnsElectBattery_py.cpp' does not exist.
Prerequisite '/dir/to/jeod/lib_jeod_Linux_7.5_x86_64/libjeod.a' of target 'build/dir/to/gunns/aspects/electrical/Batt/GunnsElectBattery_py.cpp' does not exist.
Must remake target 'build/dir/to/gunns/aspects/electrical/Batt/GunnsElectBattery_py.cpp'.`

jasonlharvey, 2025-05-02:
GUNNS has no dependencies on JEOD and vice-versa, so yes I suspect there's an unneeded dependency in your makefiles.  If you track it down to one of ours or JEOD's, let us know because we would need to fix that.

jasonlharvey, 2025-05-06:
I think you don't need to set a make argument, instead you simply need to include _gunns/lib/trick_if/S_gunns.mk_ in your sim's _S_overrides.mk_.  _S_gunns.mk_ is intended to be included in all Trick sims that use GUNNS - it ensures that all of GUNNS' various pieces are included for Trick, which is really just 2 steps: add all the code paths to _TRICK_CXXFLAGS_, and include the trickfied lib if it exists.

See our SIM_test as an example.  It will automatically include the trickfied lib if it is present, or will not include it, and let Trick compile everything, if it isn't present.  This is an example of the option thing I described above.  See  how _gunns/sims/SIM_test/S_overrides.mk_ includes _gunns/lib/trick_if/S_gunns.mk_.   _S_gunns.mk_ is the real meat: it defines GUNNS_TRICK_IF to point to the _gunns/lib/trick_if_ lib folder, and then the if/else block below it will include the lib if it exists.

Note that the way this is set up requires you to make the trickified lib separately, before building the sim, because the sim doesn't have a hard dependency on the trickified lib.  So, make the trickified lib (which you only have to do once if you're not changing gunns code), then build the sim.

Does this help?

mc1998, 2025-05-06:
It does. To summarize, the trick_if/S_gunns.mk will automatically determine whether a trickified build is present by using the `-include` statement. So first we need to build the trickified library and then link those file paths to the `TRICK_CXXFLAGS` in our sims `S_overrides.mk` file.

Trickifying before hand will produce two needed libraries. Do I also need to directly "include" the `.a` and `.o` generated files in my `S_overrides.mk`?

Have you been able to create the trickified library on your end? I navigate to `gunns/lib/trick` and type in `make`, but keep running into errors with gunns' source code.

mc1998, 2025-05-06:
We just upgraded to **Trick 19.7.3** (may be the cause) and **Gunns 19.5**. We dont have any local changes. All Im doing right now is navigating to `gunns/lib/trick` and typing `make`. The errors I see are below:
```
/home/gunns/aspects/dynamics/GunnsDynSingleRigidEom.cpp:183:10: fatal error: sim_services/Integrator/include/integrator_c_intf.h: No such file or directory
 #include "sim_services/Integrator/include/integrator_c_intf.h"
```
This file is specifically found inside trick at: `trick/include/trick/compat/sim_services/Integrator/include/integrator_c_intf.h`

I've checked both versions of trick (19.7.2 & 19.7.3) and dont see any changes in the files location.

mc1998, 2025-05-21:
Quick update @jasonlharvey! I found what was wrong. You mentioned above that step 2 was to include the code paths to `TRICK_CXXFLAGS`, however I actually needed to do this inside `GUNNS` makefile first before attempting to build the static and object libraries. (`*.a` & `*.o`)

More specifically, our `GUNNS` compiling had issues finding where our `TRICK` code lived. My fix was to include the trick paths within `gunns/lib/trick/Makefile`. More specifically, updating _line 38_ as seen below. This line adds two things: main Trick include directory and the compatibility directory for older-style includes.

`CXXFLAGS = -Wall $(OPTS) -I/home/directory/to/code/trick/include -I/home/directory/to/code/trick/include/trick/compat
`

Doing `make all` here successfully built my static library, now I can proceed with trickifying gunns. I recommend including a comment, for those who will use TRICK, to update with their respective paths for the next release.

Two examples (not limited to) where errors occur are as follows:
- `gunns/aspects/dynamics/GunnsDynSingleRigidEom.cpp` : line 183
- `gunns/core/GunnsInfraFunctions.cpp` : line 18

mc1998, 2025-05-21:
Now that I step back a bit, I understand why we would need to include the trick code paths within our `S_overrides.mk`. But this raises the question, since gunns trickification is dependent on the libraries existing, and the libraries have to be built beforehand, shouldnt a check exist that would verify a user can build the files? Just a thought

jasonlharvey, 2025-06-27:
We don't actually verify the trickified lib in our CI, and that's a gap we should close - at least then CI would be verifying that trickfied actually works, for when questions like you had about it come up.  I'll add a new ticket for that upgrade.